usableprivacy/upribox #2: Wifi isnt working - EW-7811Un

razzor1984, 2015-12-22:
Hello,
my first thought was ok my Wlan USB mini device isnt properly loaded in Raspbian Wheezy.
Quick connection through SSH and found out its supported and its runs!

upri@upribox ~ $ lsusb
Bus 001 Device 002: ID 0424:9512 Standard Microsystems Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp.
Bus 001 Device 004: ID 7392:7811 Edimax Technology Co., Ltd EW-7811Un 802.11n Wireless Adapter [Realtek RTL8188CUS]
-----------------------------------------------------------------------------------------------------------------------

upri@upribox ~ $ iwconfig
wlan0     unassociated  Nickname:"<WIFI@REALTEK>"
          Mode:Auto  Frequency=2.412 GHz  Access Point: Not-Associated
          Sensitivity:0/0
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

lo        no wireless extensions.

eth0      no wireless extensions.
---------------------------------------------------------------------------------------------------------------

im not sure but maybe i must add something in the wpa_supplicant.conf.
best regards Razzor

markushuber, 2015-12-22:
Hi, please have a look at the recommended hardware in the [README file](https://github.com/usableprivacy/upribox#list-of-recommended-hardware), getting the right WiFi hardware is tricky.
The default *hostapd* and the *RTL8188CUS* chipset do not work out of the box for access point mode. This issue is well documented e.g. in the [official raspberrypi forum](https://www.raspberrypi.org/forums/viewtopic.php?f=36&t=49793).

razzor1984, 2015-12-22:
Hi, thanks markus - it works if you this every time if yu boot up the device 
------------------------------------------------------------------------------------------------------
# to reconfigure between Wifi client and Wifi access point host:
# set static IP on wlan0 for AP; set manual for wlan0 with wpa_supplicant.conf for normal Wifi)
sudo nano /etc/network/interfaces   

# to start hostapd, uncomment # DAEMON_CONF="/etc/hostapd/hostapd.conf" )
sudo nano /etc/default/hostapd

sudo reboot
--------------------------------------------------------------------------------------------------------------------------

Luckily i searched in my hardware-box where unused things lay around and found an old 
D-Link DWA-140, with this kind of hardware its working flawless ;)
Your concerns about the raspberry pi 1 model B didn't happen.
Did a custom search on some heavy tracking pages like "The New York Times" an a bunch of other online news papers. Didn't have any lags or connection issues !! ;)  

In the feature i would be nice if the Web interface is acceptable through port:443 // https 
Its really awesome how stable it is ,great job !!!
best regards Razzor

So its working realy good at this stage of devolping !

markushuber, 2015-12-26:
Great everythings works for you @razzor1984 
Regarding the https interface, please create a new issue.
I am closing this issue regarding supported WiFi cards.

closed 2015-12-26 by markushuber