usableprivacy/upribox #3: Please add openVPN Gateway Capabilities

domci, 2016-02-07:
Hi, 

first of all: Thanks a lot for this awesome project!
I really like the upribox and using it is truely easy.

Would it be possible to use the upribox also as a VPM Gateway, 
so that all traffic will be tunneled via a VPN Service like https://nordvpn.com?

That would be awesome!

Thanks guys!

Enz9EzRU, 2016-08-07:
I second this. I`am also a NordVPN user. 

Awesome project!!!

markushuber, 2016-09-28:
Hi @domci and @Enz9EzRU , I bought myself a nordvpn account to experiment with the VPN service. Proper support in the web interface (selecting countries etc.) will not make the next release.

@domci, @Enz9EzRU Did you experiment with starting NordVPN with the openvpn command line client on the upribox yet?

markushuber, 2017-06-25:
Closing this issue for now, feature will not be implemented due to little further requests.

closed 2017-06-25 by markushuber