usableprivacy/upribox #5: Changeable VPN Port

woscholino, 2016-05-14:
Changeable VPN Port would be nice because some public wifis block specific ports.
In case the port could be adapted to 443 the problem would be solved.

markushuber, 2016-05-18:
Thankf for the request, and indeed a good option to evade firewalls of public WiFis.

woscholino, 2016-11-19:
The functionality to change the port hasn't implemented yet. Is there a way to adapt the port in a config file manually in order to change the port?

markushuber, 2016-11-19:
Hi @woscholino,

The change involves changing two config files, namely **/etc/openvpn/server.conf** and **/etc/openvpn/client_template**. Any manual changes to config files will be overridden by the automatic updates. 

I will add fix this issue until 1st of December, if you can still wait?

Thank you for your patience,
markus

woscholino, 2016-11-20:
Hey,
of course I can wait. Thank you for your support.

best regards,
woscholino

woscholino, 2016-12-04:
Hey, 
I guess the fix isn't out yet because the latest version is still 0.7. Is this correct?
I changed the port but after a while the changes were gone. 
Woscholino

markushuber, 2016-12-04:
Still working on the fix, changes are currently in a separate branch and not yet merged.

markushuber, 2016-12-06:
Hi @woscholino,

The feature to change the port of the upribox VPN server has now been rolled out, I documented the commands to change the default VPN port here: [upribox wiki - custom vpn port](https://github.com/usableprivacy/upribox/wiki/Customizations#custom-vpn-port)

Markus

markushuber, 2017-01-23:
New functionality has been added a while ago.
@woscholino if you experience any problems with setting a custom VPN port, please reopen the issue - I am closing it for now.

closed 2017-01-23 by markushuber