usableprivacy/upribox #6: Upribox on Pi3 hangs after activating Ninja mode

giantatwork, 2016-05-24:
Upribox on Pi3 hangs after activating Ninja mode.
[image]

After a reboot of the Pi3, neither _upribox_ nor _upribox-ninja_ shows up as a Wifi station.

__dmesg__ command shows errors:
```
[   29.067768] tun: Universal TUN/TAP device driver, 1.6
[   29.067781] tun: (C) 1999-2004 Max Krasnyansky <maxk@qualcomm.com>
[   29.624248] IPv6: ADDRCONF(NETDEV_UP): tun0: link is not ready
[   29.753340] IPv6: ADDRCONF(NETDEV_CHANGE): tun0: link becomes ready
[   29.846880] brcmfmac: brcmf_add_if: ERROR: netdev:wlan0 already exists
[   29.846897] brcmfmac: brcmf_add_if: ignore IF event
[   30.162788] brcmfmac: brcmf_add_if: ERROR: netdev:wlan0 already exists
[   30.162814] brcmfmac: brcmf_add_if: ignore IF event
[   30.167082] brcmfmac: brcmf_cfg80211_start_ap: SET SSID error (-52)
[   30.746363] brcmfmac: brcmf_add_if: ERROR: netdev:wlan0 already exists
[   30.746377] brcmfmac: brcmf_add_if: ignore IF event
```

markushuber, 2016-05-24:
Hi,

Thanks for reporting the issue, I am aware of this problem - as stated in the README.md of the Jessie tree - the Pi3 WiFi chip does not support multiple WiFi stations. Therefore, activating the ninja mode leads to hostapd not starting at all.

There is no easy fix for this problem, because this is a hardware limitation of the Pi3. We will however provide an additional warning into the Web Gui.

How to disable the Ninja Wifi via ssh:
`sudo /usr/local/bin/upri-config.py enable_tor no`
`sudo /usr/local/bin/upri-config.py restart_wlan`

giantatwork, 2016-05-24:
@markushuber Thanks for your fast response. Is there a way to check the type of WiFi chip the Upribox runs on? In case of a Pi3 the Ninja mode option could be disabled in the Web Gui.
If it's possible I could give it a try and create a pull request if that's ok.

markushuber, 2016-05-25:
There are many ways to check to the type of hardware chip used, in the context of the upribox sofware - I would rely on ansible to gather hardware data:
`ansible all -m setup -i /var/lib/ansible/local/environments/production/inventory_pull --connection=local`
The ansible facts include information on the hardware module used for the different network interfaces. Would be great if you created a pull request and I hope we provide development documentation soon.

commit d14cace7 (2016-07-27): Fast check for Raspberry Pi model, closes #6
closed 2016-07-27 by markushuber