usableprivacy/upribox #9: Static IP-address upribox

fozed, 2016-07-04:
Hi,

The upribox-image-v0_5 is running very well on my Raspberry Pi2 and TL-WN722N.
Using it is easy also for newby.

Would it be possible to use the upribox with a static IP-address instead of dhcp-address? and how to do that.

Thanks guys!

markushuber, 2016-07-06:
Hi @fozed ,

Setting a static IP for your upribox is possible, although we do not expose this functionality through the webinterface. 

Instructions for the static IP setup can be found here:
https://github.com/usableprivacy/upribox/wiki/Customizations#static-network-configuration

Please verify if the instructions work for you.

fozed, 2016-07-06:
Hi markus,
The instructions for the static IP did not results in the right IP, what I am doing wrong ?
In the following file my given instructions and the reaction of upribox.
[Upribox-IPtest.txt](https://github.com/usableprivacy/upribox/files/350345/Upribox-IPtest.txt)

fozed, 2016-07-07:
Hi @markushuber,

Upribox is running, after executing the instructions on the second time, on the static IP we wanted.

Thanks a lot

closed 2016-07-11 by fozed