usableprivacy/upribox #10: auto-update

Falconcrest, 2016-07-06:
Hi, 
we have a question, is planned a info in the Web interface for the auto-update? It would be immediately apparent whether the updates were executed. Your project is very interesting, similar to the project eBlocker. 
Best regards, 
Carsten & Silke.

markushuber, 2016-07-13:
Hi Carsten and Silke,

Thanks for the feedback, we will add information on the last successful update to the webinterface soon.

markushuber, 2016-07-26:
The webinterface now displays information on when the last update of the upribox software has been performed. You can find this information in the "Admin" panel: http://upri.box/more/
The displayed time is in UTC.
@Falconcrest can you verify, this is the information you have been looking for?

Falconcrest, 2016-07-26:
**@markushuber** 
Hi,
this is very nice. Exactly, so we had us envisioned it.
Great, many thanks.

Three questions:
Which tells me "version: master/3dada7c?"
Is it the number of the version update?
Can you display additionally the milestone number?

Best regards, Carsten & Silke

markushuber, 2016-07-26:
1. master/3dada7c shows the branch you currently on (master) and the last commit
2. yes it is the message-"number" of the last update
3. i replaced branch/commit with the milestone number (tag), update should reach you soon.

commit bd86b9df (2016-07-26): Tag (milestone) instead of last commit for sw version, closes #10
closed 2016-07-26 by markushuber