usableprivacy/upribox #11: Problem generating css files

tobiasdam, 2016-07-20:
Using the Jessie Image (`upribox-image-v0_5.zip (May 2016)`), each time nginx should serve `upri.css` (or rather `generic.css` or `special.css` to be exact) it fails with an `500 Internal Server Error`.

The `nginx_css_error.log` file contains following error message:

```bash
2016/07/13 21:56:02 [error] 739#0: *11 lua entry thread aborted: runtime error:
error loading module 'css' from file '/etc/nginx/lua/css.lua':
        /etc/nginx/lua/css.lua:17167: invalid escape sequence near
        '".ad,.sponsor,.panel--compare-save,#adspot-300x600'
stack traceback:
coroutine 0:
        [C]: ?
        [C]: in function 'require'
        rewrite_by_lua:3: in function <rewrite_by_lua:1>, client: 192.168.55.103,
        server: filter.upri.box,
        request: "GET /www.gamestar.de/upri.css HTTP/1.1", host: "filter.upri.box",
        referrer: "http://www.gamestar.de/"
```

It seems that the Lua file, which is loaded from https://rules.upribox.org/downloads/css.de.lua, contains several invalid escape sequences in the strings of `local specific`, for instance `\<`.

Apparently, recent versions of luajit (as used by nginx) have a stricter handling of invalid escape sequences than older versions: http://lua-users.org/lists/lua-l/2011-09/msg00267.html

Following output shows, that the nginx version, used by the jessie image, is using `luajit v5.1`:

```bash
root@upribox:/home/upri# ldd /usr/sbin/nginx | grep -i lua
        libluajit-5.1.so.2 => /usr/lib/arm-linux-gnueabihf/libluajit-5.1.so.2 (0x76db1000)
```

After some debugging with `luac` (v5.2.3) I got a working version of `css.lua`: [css.de.zip](https://github.com/usableprivacy/upribox/files/373755/css.de.zip)

markushuber, 2016-07-21:
Parsing of the CSS filters should be fixed now. Could you verify if the issue still exists? @tobiasdam

tobiasdam, 2016-07-21:
I tested the new filter files and they work like a charm. This issue is now fixed.

If someone is using the upribox in development mode and needs to update the rules manually, the easiest way is to use ssh and enter `upri-filter-update.py`.

closed 2016-07-21 by tobiasdam