usableprivacy/upribox #15: Avoid de-anonymization of Ninja Mode users

pietsch, 2016-08-20:
Are developers aware of the fact that senior Tor developers discourage people from using Tor boxes such as yours in Ninja Mode? As far as I understand, routing all connections through the same Tor chain will tell an evil Tor exit so much about this user that de-anonymizing him or her becomes a trivial task: https://lists.torproject.org/pipermail/tor-relays/2014-October/005544.html

You might be able to tackle this issue by choosing different Tor chains for each connection, and by applying various anti-fingerprinting methods, as Torbrowser does.

See also this publication by Ed Felten's group: https://www.usenix.org/conference/foci14/workshop-program/presentation/edmundson

markushuber, 2016-08-20:
Hi Christian,

We are well aware of the downsides of routing traffic through the Tor network vs. using the Tor Browser Bundle. Thank you raising this issue because its a common question we get from other tech savvy users. Also thanks for the USENIX paper; one of my favorite articles on the whole issues is this one: [Why the entire premise of Tor-enabled routers is ridiculous](http://arstechnica.com/security/2015/04/op-ed-why-the-entire-premise-of-tor-enabled-routers-is-ridiculous/).
Bottom line: for strong anonymity always use the official Tor Browser Bundle.

Therefore, we point users of the upribox to to download the Tor Browser Bundle in the user interface:
[image]

Finally, why did we even leave the Tor option in the upribox software? - protection against your ISP.
So if you suspect your ISP is tracking your browsing habits or is actively blocking some domains (e.g. the case in Austria), Tor (via the upribox Ninja mode) offers a free alternative to VPNs.

pietsch, 2016-08-20:
Hi Markus,

it's good to know you are aware of these pitfalls, and that you inform your users about it. I am sorry for opening a trouble ticket before really giving your software a try. Many thanks for the link to the Arstechnica article about Tor routers and EPICFAIL – it's a good read!

By the way, Tor Browser Bundle was renamed to Tor Browser in 2014. You might also want to tell your users that the [Tails](https://tails.boum.org/) live OS is generally considered the safest way to run the Tor Browser.

I am hesitating to close this ticket. Perhaps you can indeed borrow some ideas from the Tor Browser to avoid EPICFAIL?

markushuber, 2016-08-22:
I created #16 to fix the link to Tor Browser and in addition include a link to Tails. Thank you for the suggestions.

To answer the second part of your question: *borrow some ideas from the Tor Browser to avoid EPICFAIL*:
1) There are functionalities of the Tor Browser which will never work on a network-level solution. For example, imagine you load *https://somewebsitewhichisusingtls.com/fingerprint.js* over a torrified network vs. the Tor Browser: on a network level you cannot block the fingerprinting script without a MiTM attack on the TLS connection, which is not an option at all or alternatively block the entire domain (imagine if we started to block google.com etc.).
2) EPICFAIL et al.: If your threat model includes state-sponsored agencies you had to take a number of additional protections steps in addition to even Tor Browser, see e.g. [Best practices for Tor us](http://security.stackexchange.com/questions/43369/best-practices-for-tor-use-in-light-of-released-nsa-slides). So once again our threat model includes censorship and avoiding monitoring by your ISP.

markushuber, 2016-08-25:
@pietsch Added link to Tails to the user interface and changed Tor Browser Bundle to Tor Browser. Hope to have answered your questions, I will close this issue for now. Thank you again for raising the question, this issue will serve as an import starting point for future similar questions.

closed 2016-08-25 by markushuber