usableprivacy/upribox #21: Wifi SSID's available but no Internet connections

jziggy101, 2016-09-10:
I just flashed my sdcard with a fresh copy of Upribox and am running it on Raspberry PI 2 with a TP-Link TL-WN722N.  I can turn on both the 'upribox' & 'ninja-upribox' and connect to them both.  I can access the "upri.box" site but neither SSID's can access the internet. I am running v0.6-45-g27d2bda.  I had originally installed your firmware back in June and both SSID's worked and I was able to use TOR on ninja-upribox.  The box has been online for 2 days and I can still not reach the internet on any device while connected to either SSID.  Any ideas?

markushuber, 2016-09-11:
Hi @jziggy101,

It sounds like the setup is working but your box does not have a valid Internet connection. Please connect to the upribox WiFi and verify the following:
* can you ping sites, e.g. is `ping 8.8.8.8` -> yes, the upribox is connected to the Internet
* is there any difference between *http* and *https* sites? -> is e.g. https://github.com working but http://www.cnet.com/ fails to load?

best,
markus

jziggy101, 2016-09-11:
I disabled the ninja SSID hoping this might clear things up but nothing changed after a reboot.  
I can ping 8.8.8.8 successfully. I cannot get to google.com, http://www.google.com or https://www.google.com msg reads server DNS adresss could not be found.  My pc's and tablet indicate that there is a connection to the SSID but no internet available.  I even tried to connect to google.com via an ip address of 216.58.216.238 with the same message results.

markushuber, 2016-09-11:
This seems like a DNS issue, which seems strange as you can resolve/ping `upri.box` otherwise you could not have have used the webinterface?

jziggy101, 2016-09-11:
I noticed it also took several attempts to get to the upri.box site on the box.  
Now thinking about it the upribox has to have access to the net because I flashed the May 2016 image to the sdcard and then left the unit online so that it could get the updates for the firmware which it did.  My previous sdcard was created in July 2016 and worked flawlessly until last week when it stopped connection to any device I tried to attach to the SSIDs.  I figured the sdcard must have developed an issue and I used a new class 10 card to flash the image and start over.  Not really sure whats going on now.  I can connect a device to either the silent or ninja SSID but nothing is able to access the internet.

jziggy101, 2016-09-11:
Not sure if anything here wiill help.

upri@upribox:~ $ ifconfig
eth0      Link encap:Ethernet  HWaddr b8:27:eb:f5:01:f5
          inet addr:192.168.1.18  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: 2601:403:0:7bf0:23c9:a16f:9980:e2a4/64 Scope:Global
          inet6 addr: fe80::b940:ae08:8612:657a/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:69008 errors:0 dropped:8 overruns:0 frame:0
          TX packets:2760 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:7057937 (6.7 MiB)  TX bytes:347909 (339.7 KiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:6136 errors:0 dropped:0 overruns:0 frame:0
          TX packets:6136 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:481252 (469.9 KiB)  TX bytes:481252 (469.9 KiB)

tun0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
          inet addr:192.168.155.1  P-t-P:192.168.155.2  Mask:255.255.255.255
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:10000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

wlan0     Link encap:Ethernet  HWaddr 02:de:27:1d:1f:30
          inet addr:192.168.55.254  Bcast:192.168.55.255  Mask:255.255.255.0
          inet6 addr: fe80::3e0c:2038:5504:3f9/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:11830 errors:0 dropped:0 overruns:0 frame:0
          TX packets:13117 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:918178 (896.6 KiB)  TX bytes:1642686 (1.5 MiB)

markushuber, 2016-09-12:
Thanks for posting your network config @jziggy101 , I am still puzzled why your upribox seems to have connection issues and one thing I noticed: your box has an IPv6 address.

Currently we do not support IPv6 fully, could you therefor please try to disable IPv6 to see if this changes anything? Add the following lines to the end of the **/etc/sysctl.conf** file on the upribox:
`net.ipv6.conf.all.disable_ipv6 = 1`
`net.ipv6.conf.default.disable_ipv6 = 1`
`net.ipv6.conf.lo.disable_ipv6 = 1`

Then apply the changes by running **sudo sysctl -p**.

If this does not help, I will write you a list of files that would help me to debug and fix the issue.

jziggy101, 2016-09-12:
Made the proposed changes and so far.
-can't reach upri.box   server DNS address could not be found
-can't reach goog.com, htpp://www.google.com, https://www.google.com

I can ping 8.8.8.8 but I cannot ping several other ip's that I own on the internet.

ipv6 addresses are no longer listed during an ifconfig

markushuber, 2016-09-12:
Thanks for trying out the things - IPv6 has not been the issue, so back to DNS. 
Can you verify if the DNS service is running? 
Could you please post the output of `sudo service dnsmasq status` and `sudo journalctl -u dnsmasq.service`?

jziggy101, 2016-09-13:
root@upribox:/home/upri# service dnsmasq status
● dnsmasq.service - dnsmasq - A lightweight DHCP and caching DNS server
   Loaded: loaded (/lib/systemd/system/dnsmasq.service; enabled)
  Drop-In: /run/systemd/generator/dnsmasq.service.d
           └─50-dnsmasq-$named.conf, 50-insserv.conf-$named.conf
   Active: failed (Result: exit-code) since Tue 2016-09-13 12:02:28 UTC; 2min 32s ago
  Process: 542 ExecStart=/etc/init.d/dnsmasq systemd-exec (code=exited, status=3)
  Process: 369 ExecStartPre=/usr/sbin/dnsmasq --test (code=exited, status=0/SUCCESS)

Sep 13 12:02:24 upribox dnsmasq[369]: dnsmasq: syntax check OK.
Sep 13 12:02:28 upribox dnsmasq[542]: dnsmasq: cannot open log /var/tmp/log...ry
Sep 13 12:02:28 upribox dnsmasq[542]: cannot open log /var/tmp/log/dnsmasq/...ry
Sep 13 12:02:28 upribox systemd[1]: dnsmasq.service: control process exited...=3
Sep 13 12:02:28 upribox systemd[1]: Failed to start dnsmasq - A lightweight...r.
Sep 13 12:02:29 upribox systemd[1]: Unit dnsmasq.service entered failed state.
Hint: Some lines were ellipsized, use -l to show in full.
root@upribox:/home/upri#

root@upribox:/home/upri# sudo journalctl -u dnsmasq.service
-- Logs begin at Tue 2016-09-13 12:02:08 UTC, end at Tue 2016-09-13 12:05:39 UTC
Sep 13 12:02:16 upribox systemd[1]: Starting dnsmasq - A lightweight DHCP and ca
Sep 13 12:02:24 upribox dnsmasq[369]: dnsmasq: syntax check OK.
Sep 13 12:02:28 upribox dnsmasq[542]: dnsmasq: cannot open log /var/tmp/log/dnsm
Sep 13 12:02:28 upribox dnsmasq[542]: cannot open log /var/tmp/log/dnsmasq/dnsma
Sep 13 12:02:28 upribox systemd[1]: dnsmasq.service: control process exited, cod
Sep 13 12:02:28 upribox systemd[1]: Failed to start dnsmasq - A lightweight DHCP
Sep 13 12:02:29 upribox systemd[1]: Unit dnsmasq.service entered failed state.

markushuber, 2016-09-13:
Thanks, @jziggy101, the output points to a bug related to the latest changes from #20.
Does the directory */var/tmp/log/dnsmasq/* exist on your box and what are its permissions? (`ls -lah /var/tmp/log/dnsmasq/`), and what are the permissions of *ls -lah /var/tmp/log/dnsmasq/dnsmasq.log`?

jziggy101, 2016-09-13:
no directory at /var/tmp/log/dnsmasq but log file is in /var/tmp/log

root@upribox:/var/tmp/log# ls
dnsmasq_ninja.log           nginx_error.log             rqworker.err.log
nginx_access.log            nginx_interface_access.log  rqworker.out.log
nginx_blackhole_access.log  nginx_interface_error.log   rsyslog
nginx_blackhole_error.log   openvpn.log                 supervisord.log
nginx_css_access.log        openvpn-status.log
nginx_css_error.log         privoxy

root@upribox:/var/tmp/log# ls -lah /var/tmp/log/
total 24K
drwxrwx--x 4 root    tmp-logger  360 Sep 13 12:02 .
drwxrwxrwt 3 root    root       4.0K Sep  9 20:11 ..
-rw-r----- 1 dnsmasq root        511 Sep 13 12:02 dnsmasq_ninja.log
-rw-r--r-- 1 root    root          0 Sep 13 12:02 nginx_access.log
-rw-r--r-- 1 root    root          0 Sep 13 12:02 nginx_blackhole_access.log
-rw-r--r-- 1 root    root          0 Sep 13 12:02 nginx_blackhole_error.log
-rw-r--r-- 1 root    root          0 Sep 13 12:02 nginx_css_access.log
-rw-r--r-- 1 root    root          0 Sep 13 12:02 nginx_css_error.log
-rw-r--r-- 1 root    root          0 Sep 13 12:02 nginx_error.log
-rw-r--r-- 1 root    root          0 Sep 13 12:02 nginx_interface_access.log
-rw-r--r-- 1 root    root          0 Sep 13 12:02 nginx_interface_error.log
-rw-rw-r-- 1 openvpn openvpn    3.3K Sep 13 12:02 openvpn.log
-rw------- 1 openvpn openvpn     232 Sep 13 12:41 openvpn-status.log
drwxr-xr-x 2 privoxy root         60 Sep 13 12:02 privoxy
-rw-r--r-- 1 root    root          0 Sep 13 12:02 rqworker.err.log
-rw-r--r-- 1 root    root        153 Sep 13 12:03 rqworker.out.log
drwxr-xr-x 2 root    root        160 Sep 13 12:02 rsyslog
-rw-r--r-- 1 root    root        605 Sep 13 12:02 supervisord.log
root@upribox:/var/tmp/log#

at present there is no ninja SSID enabled on this upribox as I disabled it hoping it might clear the problem.

markushuber, 2016-09-13:
I can not reproduce the bug right now and this should fix the issue:

`sudo mkdir /var/tmp/log/dnsmasq`
`sudo chown dnsmasq /var/tmp/log/dnsmasq`
`sudo service dnsmasq start`

jziggy101, 2016-09-13:
made the changes.
I can now reach the upri.box reliably

version v0.6-45-g27d2bda

Internet connectivity via dns appears to be working.  google.com, yahoo.com, github.com all working.

markushuber, 2016-09-13:
Thank you ever so much @jziggy101, you helped us to discover a serious bug!
Will provide a bug-fix asap.

jziggy101, 2016-09-13:
Thanks for all your help Markus.  I truly enjoy using this firmware.  
I gave a talk on it to a local user group in early August and about a dozen users have told me they installed it and were using it in August.  I hope the bug fix will keep them using it.  I have not heard from any that were having problems but they are a large user group so I might hear from them on the next meeting later this month.
Please keep up the great work.

jziggy101, 2016-09-13:
I re-enabled the ninja SSID as this is a Raspberry 2 and I cannot connect to the SSID.  Not a problem but I did want to see if that was working again.  I have disabled it again and sient is still working fine.

markushuber, 2016-09-14:
@jziggy101 I committed a fix for the initial issue you stated yesterday, can you confirm that the network connectivity issue is fixed and the fix works after reboots?
If yes, I will close that issue.
In case the ninja Wifi issue persists, could you please open a separate issue?

Thank you for helping us important issues.

jziggy101, 2016-09-14:
I am on v0.6-48-gd5fb9ea and the silent SSID connectivity is working great.

I have the ninja SSID disabled at the moment but I can check to see if it has any issues and start a new issue should that be the case.
Thanks

closed 2016-09-14 by jziggy101