usableprivacy/upribox #25: Ninja-upribox wireless SSID no connectivity

jziggy101, 2016-09-14:
I enabled the Ninja-upribox SSID and neither my laptop or tablet can connect to the SSID.  My devices will attempt connection, attempt to resolve an ip address  then fail and start the process again.
Sometimes the laptop will connect with only a 169.255.x.x address and no network connectivity nor internet.
Back in August I had used the ninja SSID with great success, it connected and I had access to the TOR network via that connection.  Sometime late in August I was no longer able to connect to the ninja SSID.

tun0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
          inet addr:192.168.155.1  P-t-P:192.168.155.2  Mask:255.255.255.255
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:10000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

wlan0     Link encap:Ethernet  HWaddr 02:de:27:1d:1f:30
          inet addr:192.168.55.254  Bcast:192.168.55.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:24886 errors:0 dropped:0 overruns:0 frame:0
          TX packets:30589 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:4129706 (3.9 MiB)  TX bytes:23487945 (22.3 MiB)

wlan0_0   Link encap:Ethernet  HWaddr 02:de:27:1d:1f:31
          inet addr:192.168.56.254  Bcast:192.168.56.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:518 errors:0 dropped:12 overruns:0 frame:0
          TX packets:569 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:70800 (69.1 KiB)  TX bytes:94564 (92.3 KiB)

root@upribox:/home/upri# iwconfig
tun0      no wireless extensions.

wlan0     IEEE 802.11bgn  Mode:Master  Tx-Power=20 dBm
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off

lo        no wireless extensions.

wlan0_0   IEEE 802.11bgn  Mode:Master  Tx-Power=20 dBm
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off

eth0      no wireless extensions.

markushuber, 2016-09-16:
Hi @jziggy101, so the tissue with the ninja Wifi is - that you never get a valid IP address for the WiFi network? 
For me the ninja WiFi works with a Pi2 and the TPLink dongle, I verified everything works with a Linux Desktop, and an Android Smartphone.

If you connect successfully to the ninja WiFi your devices should get an IP starting with *192.168.56.*. Which clients/devices are you using?

jziggy101, 2016-09-16:
Markus I understand how it did work until about the same time my other ssid
problem started.  I used to get a 192.168.56.xx ip adess but currently I am
not even get a connection to the ssid.  What ever device I try to connect
with that ninja ssid never negiotiates an ip adress so no connection.  I
was using the ninja ssid & connecting through the tor network easilyand
then suddenly one day nothing would work.

jziggy101, 2016-09-17:
Markus I see a new version on my upribox so I tested the ninja ssid and it connected, received ip and is on the tor network.

closed 2016-09-17 by jziggy101

markushuber, 2016-09-17:
Great to hear, thanks for reporting issues back to us - very much appreciated.