usableprivacy/upribox #30: Updates

jpbudinger, 2016-09-29:
I cannot access the internet and updates don't seem to be coming every 4 hours.  The latest distribution is still 0.5 from May 2016.  Why isn't latest updated image available.?
Thanks
John

markushuber, 2016-09-29:
Hi @jpbudinger, just released the latest image, please give it a try:
https://github.com/usableprivacy/upribox/releases/latest

your feedback is appreciated.
markus

jpbudinger, 2016-09-29:
Thanks Marcus

jziggy101, 2016-09-29:
Markus I loaded the v0.7 firmware to a new sdcard and installed it into my Pi2.  Should the version number at the bottom of the admin page still show v0.6 instead of v0.7?  Currently its showing v0.6-55-g745986a

I am a friend of John's and was trying to help him with his issues yesterday.

markushuber, 2016-09-29:
The new version number v0.7 should show up in 4h latest.

jpbudinger, 2016-09-29:
Duh!  I think I am a bonehead.  Through all of this, I have been 
changing WiFi dongles, software, etc. and never remembered to plug in 
the Cat5 cable.  All is working fine at this time.

closed 2016-09-29 by markushuber

jpbudinger, 2016-09-29:
Sorry for all the trouble Marcus (and jziggy101).  Probably a good thing to mention to people about the cable.  Maybe I should go back to bed.  Thanks for all your help.

jziggy101, 2016-09-29:
version did finally change to v0.7-1-g1f99a5f

My network provides IPV6 and previously you had mentioned you didn't completely support IPV6.  Any changes on that?  my ifconfig shows ipv6 addresses assigned to both the silent & ninja SSID's.  I have internet access on both silent & ninja as well as TOR access on ninja.