usableprivacy/upribox #33: .box TLD official

schmiedecker, 2016-11-19:
The TLD .box will become routable anytime soon, I'd recommend to change the local admin interface to https://my.upribox instead.

See also https://www.heise.de/newsticker/meldung/Neue-Top-Level-Domain-box-bringt-manche-Netze-durcheinander-3491185.html

markushuber, 2016-11-19:
Thanks @schmiedecker,

This is currently not an issue because the upribox does not use a global DNS service to resolve **upri.box**.

markus

closed 2016-11-19 by markushuber

schmiedecker, 2016-11-19:
Sure, but any admin with a misconfigured (external) DNS resolver will land
on the upcoming upri.box domain. Copy the interface, ask for the password,
done.

markushuber, 2016-11-20:
The upribox intercepts and modifies any DNS request, so our current solution works independent of the used external DNS service. Thanks for bringing the issue up.