usableprivacy/upribox #35: DHCP Server

woscholino, 2016-11-23:
Hey,

just wanted to try the new zero-config. Unfortunately I found out that the ARP spoofing does not work with my Speedport W 724v. Other products are using ARP spoofing as well e.g. eblocker and their forum says in order to use ARP spoofing with the Speedport W 724v the DHCP server of the router has to be deactivated.
I am quite sure that I am not the only one in Germany that is a telekom customer and is using this router.
In case you have a other solutions in order to get the ARP spoofing working I would appreciate.

Best regards,
woscholino

markushuber, 2016-11-24:
Hello @woscholino,

It seems that the issue arises from IPV4/IPV6 Dualstack setups. Can you confirm that your Internet access also provides IPV6 access? The prob. easiest way is to check with Google's test: (http://ipv6test.google.com/)[http://ipv6test.google.com/] .

markus

woscholino, 2016-11-24:
Yes, I can confirm that my intern access provides both IPv4 and IPv6.

markushuber, 2016-11-24:
Thanks @woscholino, this issues is fixable and requires some effort in order to upgrade the upribox software to full IPv6 compatibility. Except this to be fixed in the next couple of month.

markushuber, 2017-03-15:
@woscholino @tobiasdam implemented IPv6 support for the apate zero-config daemon. Your box should receive the update today. Could try to enable apate in your network and let as know if it works for you?

markushuber, 2017-03-30:
Closing this issue for now, as our *apate* zero-config daemon now supports IPv6.

closed 2017-03-30 by markushuber

woscholino, 2017-04-19:
Activating DHCP Server and apate zero-config daemon works for me.
Thank you.

woscholino, 2017-04-19:
Unfortunately I have to correct my answer. It works on my MacBook with disabled IPv6. Both my iPad and iPhone get an IPv6 address and therefore nothing is filtered.

markushuber, 2017-04-19:
**apate** is up and running and nothing gets filtered on your iPad/iPhone?
We tested our IPv6 feature mainly with Android devices and at least for us it worked.

woscholino, 2017-04-21:
Unfortunately yes. I deactivated the dhcp server of my speedport router but it still gives IPv6 addresses. Maybe this is the problem.

markushuber, 2017-04-22:
Too bad, I experimented with the dual-stack setup from my provider for some weeks:
* No option available to disable/configure IPv6 on the modem
In the end, called my provider (UPC Austria) to switch me back to IPv4 (set "IPv4 only" flag). The main reason for me was the lack to access my services while on the road (IPv4 port forwarding does not work with my provider if IPv6 is enabled).

woscholino, 2017-04-23:
Unfortunately the Telekom does not disable dual-stack and my router does not provide any option for deactivating IPv6. That means I have live with an additional WLAN.

markushuber, 2017-04-23:
Yep that's unfortunate, if you have an additional router you could disable IPv6 there and route all traffic through your upribox.