usableprivacy/upribox #36: Whitelist

woscholino, 2016-11-23:
Hey there,

a whitelist would be great because some websites block the access to the website in case a adblocker is active.

best regards,
woscholino

markushuber, 2016-11-24:
Hi @woscholino,

There are indeed some websites which detect adblockers and stop working. We will push fixes for those websites - could you provide a list of websites which create problems for you?

best,
markus

woscholino, 2016-11-24:
The first website I have troubles with is:
sueddeutsche.de

Best regards,
woscholino

markushuber, 2016-11-24:
I am relieved you mentioned a quality paper such as sueddeutsche, was afraid you were going to respond *bild.de* ;-)

There are a number of websites that require workarounds/whitelisting. I will keep this issue open to discuss progress on our whitelisting.

intrepid-ibex, 2016-12-01:
Hi,

hier sind keine Downloads bei aktiver UPRIBOX möglich :

http://de.downmagaz.com/

#################################

hier kommt sporadisch noch Werbung durch :

http://www.ariva.de/goldpreis-gold-kurs/realtime-chart?boerse_id=130

oder die Meldung : Verbindung fehlgeschlagen

Best regards 

intrepid-ibex

markushuber, 2016-12-02:
@woscholino fix for sueddeutsche.de has been activated, could you verify that the white-listing works for you?

@intrepid-ibex bei dem ersten Link ist das Problem bei uploaded.net nicht bei der ersten Seite, werden wir uns ansehen. Bei der zweiten Seite konnte ich auf einen ersten Blick keine Probleme erkennen. Kannst du ein neues Issue aufmachen? - hier würde ich gerne Adblock Erkennung diskutieren.

woscholino, 2016-12-02:
Thank you, sueddeutsche.de works fine for me.

intrepid-ibex, 2016-12-03:
Hi,

sueddeutsche. de funktioniert jetzt.
klausminator.de funktioniert jetzt.
robinson.com funktioniert jetzt.
bild.de funktioniert noch nicht :-( 
mydealz.de macht noch Probleme

Besteht die Möglichkeit die "whitelist" selbst zu ändern bzw. eine "UserWhitelist" zu führen?

LG intrepid-ibex

markushuber, 2016-12-03:
Momentan gibt es keine Möglichkeit selbständig Ausnahmen zu definieren. Neuigkeiten zu Whitelisting werde ich hier posten.

mannaz, 2017-07-24:
Es werden immer wieder cloudfront.net domains geblockt. 

Z.B. hier https://bitbucket.org: d301sr5gafysq2.cloudfront.net

markushuber, 2017-07-27:
hej @mannaz danke für das finding, wir haben da ein overblocking der gesamten cloudfront.net tld-domain, da auf cloudfront auch einige Tracker gehostet werden. Regelset sollte in den nächsten Tagen gefixed sein.

intrepid-ibex, 2017-07-28:
https://www.mydealz.de  : der Aufruf über die grüne Schaltfläche ( zum Deal ) wird geblockt. Der Bildschirm bleibt "Leer" . 
MfG

markushuber, 2017-08-04:
@intrepid-ibex the other day, I had a quick look at the issue. mydealz.de: does not work because of a more general problem: **referal-networks/sites**. We are currently any kind of referral (tracking network). Going to whitelist the affected services in the next days.

intrepid-ibex, 2017-08-04:
Das Verhalten der UPRIBOX ( Silentmodus = on, APATE = ON ) ( Endgerät = Windows 10 64Bit, Firefox 64bit ) ist unterschiedlich :

1. Anbindung per WLAN : bild.de wird geblockt, Hinweis auf Adblocker wird angezeigt

2. Anbindung per Ethernet :  bild.de wird korrekt angezeigt.

Andererseits stellen wir fest, dass bei einer bestimmten WEBSeite ( ariva.de ) die Werbeblockung 
bei WLAN Anbindung zuverlässiger funktioniert als bei Ethernet Anbindung.
Falls noch zusätzliche Informationen benötigt werden, bitte melden.

MfG

markushuber, 2017-12-01:
This issue is now open for quite some time, let me give you an update on the discussions we had in the upribox Team:

### Adblock-Detection
As things stand now we will not actively participate in the anti-adblock armsrace, meaning effectively:
if a website (like bild.de) blocks access because of the upribox, you have two options:
**A) Stop using websites like bild.de** ;-)
**B) Temp. deactivate your the upribox for a certain device (v2 feature)**

### Affiliate Networks
Referrals from daily-deals sites, Google search, price comparison sites etc. track you. That's also why we block affiliate networks. Technically speaking we only have two options here: completely block them or whitelist affiliate networks. Options we discussed here:
**A) Temp. deactivate your upribox for a certain device**
**B) A webinterface-option for the upribox to whitelist all affiliate networks 
(warning users about the loss of privacy)**

@intrepid-ibex what do you think about the potential new option?