usableprivacy/upribox #37: Documentation

markushuber, 2016-11-24:
We put a lot of thought into the design of the upribox Software but have documented our design.
A proper documentation will help to explain and communicate the *security*, *privacy* and *networking* features of the upribox Software.
We are considering to use Sphinx, other suggestions are very welcome.

ThorstenS-linux, 2016-11-29:
I´m also using sphinx with a slightly modified rtd-theme. Its perfect for man/html/pdf/epub output.
(especially manpages are super cool 👍 ).  
I think to go with sphinx is a good decision!

markushuber, 2016-12-10:
upribox documentation will be hosted on readthedocs.org: 
[image]

commit bd0bfc94 (2017-07-12): Merge pull request #79 from usableprivacy/argparse

Argparse and documentation, closes #37
closed 2017-07-12 by markushuber