usableprivacy/upribox #44: Statistics do not work

cmalzer2, 2017-01-22:
After login to upri.box gui and changing to statistics I get only the circling circle. No statistics since a couple weeks. Logged in on mac.sierra with Opera and Safari - same result!
[image]

markushuber, 2017-01-23:
Hi @cmalzer2,

Could please post the Software Version your box is running? (in the Admin menu)?

markus

cmalzer2, 2017-02-04:
Hallo Markus,
 I use v0.7-26-g2fc2264
Christoph

markushuber, 2017-02-05:
Hi Christoph,

You are already using the latest Software version - which seems good. We could however not reproduce the error on our test devices.

Have you tried rebooting your box to see if the problem still remains? (either via ssh or simply by un-plugging the power supply)

best Markus

cmalzer2, 2017-02-11:
Hi Markus,
I do regular restarts due the great reliability of UPC/Vienna;-)
Anyhow I just did a restart via ssh, no change. I can give a Teamviewer remote access if you want!
Regards Christoph

markushuber, 2017-02-13:
Hi Christoph, please send me an E-Mail: markus AT upribox DOT org
I am sure we find a suitable way for debugging, esp. because you live in Vienna as well. ;-)

m

jziggy101, 2017-03-09:
Just stopped in to see if anything was new and saw this issue.  I had not tried to access the statistics for some time but it turns out I have the same issue.  Rotating circle and no statistics ever show.

I am on v.07-38-g00fc46f

markushuber, 2017-03-09:
Thanks to @cmalzer2 we located the issue. We are currently testing a solution which should available to all upribox users, next week.

markushuber, 2017-03-14:
commit #48 fixed the issue, statistics are responsive and displayed again.

markushuber, 2017-03-15:
@cmalzer2 is still experiencing some issues, @jziggy101 does our fix work for you?

markushuber, 2017-03-16:
issue is resolved, thanks @julianschritt for the fix.

closed 2017-03-16 by markushuber