usableprivacy/upribox #50: Request: Web interface accessibility via the IP

boolean-is-null, 2017-03-18:
Hello and a massive thank you for this project!

I'd love to access the web interface from devices on the same network as the upribox, but not connected to the WiFi(s).

So basically I'd like to access the web interface via the IP, not the host name `upri.box`.

When I access the upribox directly via the IP I get a blank page. Source:

```
<html>
<head>
<script type="text/javascript">parent==window&&opener&&close()</script>
</head>
<body style="background:transparent">
</body>
</html>
```

Is that an issue on my side or is that just not a feature implemented yet?

boolean-is-null, 2017-03-20:
I don't know if I'm clear enough, but I'm sure this super well elaborated illustration will help explain the network setup.

I'd like to access the UpiBox's web interface, via the IP from Client 1, which is not directly connected to the UpriBox.

[image]

markushuber, 2017-03-21:
Hi @boolean-is-null wow thanks for the great illustration. Well basically you have two options here:
* if you want to protect *Client 1* with the upribox, you can enable apate and re-route all traffic through the upribox. As a side-effect you can also reach the webinterface via *Client 1*
* if you want to leave your set-up as is, the easiest way would be to make an entry in the *hostfile* of *Client 1* pointing to upri.box and its IP. This step differs based on the operating system you are using.

boolean-is-null, 2017-03-21:
Hey!

Alright, yeah, I guess that would work!

I was looking at the nginx configuration and from what I can tell it only listens to the `servername upri.box`. I believe that's what why I can't access the web interface from `client 1`, would a configuration adjustment for nginx not also be a viable solution?

markushuber, 2017-03-24:
Rehi!

Good catch with the nginx configuration :-)

nginx however listens to `upri.box` for a reason:
all blacklisted/filtered domain are redirect to the `nginx` instance. In addition to responding with an empty domain, we also try to reset any previous tracking cookies with `nginx`. Therefore, we cannot make the web interface the default for all domains (the webinterface would then be shown for ad/tracking domains).

boolean-is-null, 2017-03-28:
That makes sense. thanks for the elaboration!

Hosts file adjustment it is then.

I didn't find anything about the topic and I believe it's nonsense to post another issue for that, but is it okay/advised to manually update the raspbian?

markushuber, 2017-03-28:
I am closing this issues for now.
Regardings updates: security updates are automatically installed on the upribox. We do not test other package updates for unwanted site-effects.

closed 2017-03-28 by markushuber