usableprivacy/upribox #52: Cannot connect to Wifi AP

mor3dr3ad, 2017-04-01:
Hey there,

I have installed upribox on a RPi 2 Model B and it is connected to the router. It has internet connection and I can access via ssh. However, I cannot (from my linux machine) connect to the upribox wifi - it will not accept the password (which I have gone and changed within wlan.facts and therefore: yes, I am sure I am using the right passwd). 

However, I have to say I am not using one of the supported wifi-sticks: Edimax with the EW-7811Un chipset. However, as the upribox wifi shows up I suppose it should at least work as an access point.

somebody have similar issues?

And Markus: keep up the great work!

tobiasdam, 2017-04-01:
Hey,

if you manually adjusted the `wlan.fact` file, it might be possible that the changes had no been applied by the time you tried to authenticate.
Also hostapd might not accept your supplied password, if it does not consist of at least `8` up to `63` characters or contains some special characters.

Could you please try to change the password via the webinterface or  `upri_config.py set_password "<password>"`?

mor3dr3ad, 2017-04-01:
Hey there, 
thank you. I saved the file and restarted the Wlan via upri-config.py restart_Wlan; the password has more than 8 char. 

Setting it via upri-config.py set_password prompts for special chars and digits - still does not work. I suspect it might have to do with the Wlan stick. I will get a new one Mon/Tue and will post again if it works then.

markushuber, 2017-04-03:
Hi @mor3dr3ad,

Once you manually set the password, you have to run `upri_config.py restart_wlan` in order to use the password. In order to further help you, just to be clear:
* You can see the WiFi but connections fail? (authentication failed)

best markus

P.S. Thanks for the flowers and for the record: @tobiasdam and @julianschritt do most of the awesome work for our upribox project.

mor3dr3ad, 2017-04-03:
Hey Markus,

yes that is right - I can see the upribox Wifi. The RPi also has inet access and I can reach it via ssh.

However, I cannot connect to the Wifi; my OS (Ubuntu 16.04) just keeps asking me for the passwd. Btw this was true for the default passwd *and* the changed passwd. 

Like I said, I will get a supported Wifi-Stick today or tomorrow anyways. However, if - for troubleshooting or wanting to add the rather popular (for RPi in general) Edimax Stick to your supported devices - you need some logs from my side, let me know.

I'd be glad to somehow contribute to your project.

markushuber, 2017-04-04:
Getting the right hardware is tricky and we recommend to use the TP-Link TL-WN722N USB dongle.
Let me know if changing the WiFi dongle worked for you. BTW, depending on the Edimax Stick - not all versions of this brand support Access Point mode out-of-the-box.

mor3dr3ad, 2017-04-04:
The TP-Link does not work at all - it is not even recognized. Same issue on my linux machine though so maybe it is faulty. Or are there any additional steps to be followed?

markushuber, 2017-05-10:
Have you been able to resolve the WiFi issue @mor3dr3ad ?
You last update suggested that your Wifi dongle was faulty.

mor3dr3ad, 2017-05-17:
Hey, just got a new dongle today. However, still the same issue: RPi will not recognize the dongle out of the box. lsusb will show the vendor id: 
`Bus 001 Device 004: ID 2357:010c  
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`

but does not show anything else. Any suggestions on how to proceed? Is there a driver I need to install?

markushuber, 2017-05-19:
Hi @mor3dr3ad,

If you perform a `sudo ifconfig` does your wifi dongle show up as **wlan0** or e.g. **wlan1** ?

mor3dr3ad, 2017-05-21:
Hi @markushuber,

neither nor. The stick is just not recognized. possible Kernel/driver compatibility issue?

mor3dr3ad, 2017-05-24:
So, I decided to switch RPi's and am running upribox on a RPi3 now instead of a RPi 2 with the TP-Link. Works fine with the on-board Wifi, however, the WN722N USB dongle did not work for me.

markushuber, 2017-05-29:
The issue is still strange and I would be happy to resolve to the issue - in case anyone else runs into similar issues. Could you post the output of the `dmesg` command after you connect the TP-Link dongle?

mor3dr3ad, 2017-06-01:
So this is what dmesg said right after pluggin in the stick. This seems (to me at least) to be okay. 

`[ 3734.407603] usb 1-1.4: new high-speed USB device number 4 using dwc_otg
[ 3734.548454] usb 1-1.4: New USB device found, idVendor=2357, idProduct=010c
[ 3734.548468] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 3734.548477] usb 1-1.4: Product: 802.11n NIC
[ 3734.548484] usb 1-1.4: Manufacturer: Realtek
[ 3734.548491] usb 1-1.4: SerialNumber: 00E04C0001`

However, sometime later this line pops up:

`IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready`

I remember that I had to disable IPv6 on other RPi installations - might this have sth to do with it?

markushuber, 2017-06-06:
It seems that you bought a new revision of the TP-Link WN722N USB dongle (**version 2.0**). This version uses a chipset which is not suitable for WiFi access points out of the box. We will update the documentation accordingly. For information on the different hardware versions: [wikidevi](https://wikidevi.com/wiki/TP-LINK_TL-WN722N).

markushuber, 2017-06-06:
Updated the documentation. Closing the issue for now.

closed 2017-06-06 by markushuber