usableprivacy/upribox #86: Only download security updates with unattended upgrades

markushuber, 2017-09-21:
Currently *unattended-upgrades* is configured to automatically download updated packages. This option should be disabled while keeping the setting to automatically download and install security updates.

tobiasdam, 2017-10-04:
Apparently, this option is already disabled:
```
// Do "apt-get upgrade --download-only" every n-days (0=disable)
APT::Periodic::Download-Upgradeable-Packages "0";
```

The unattended-upgrades package  in the raspbian repositories seems to contain the default debian configuration `"origin=Debian,codename=${distro_codename},label=Debian-Security"`, which does not target any valid raspbian repository.

Available repositories are:

``` 
root@upribox:/home/upri# apt-cache policy
Package files:
 100 /var/lib/dpkg/status
     release a=now
 500 http://archive.raspberrypi.org/debian stretch/ui armhf Packages
     release o=Raspberry Pi Foundation,a=testing,n=stretch,l=Raspberry Pi Foundation,c=ui,b=armhf
     origin archive.raspberrypi.org
 500 http://archive.raspberrypi.org/debian stretch/main armhf Packages
     release o=Raspberry Pi Foundation,a=testing,n=stretch,l=Raspberry Pi Foundation,c=main,b=armhf
     origin archive.raspberrypi.org
 500 http://mirrordirector.raspbian.org/raspbian stretch/rpi armhf Packages
     release o=Raspbian,a=stable,n=stretch,l=Raspbian,c=rpi,b=armhf
     origin mirrordirector.raspbian.org
 500 http://mirrordirector.raspbian.org/raspbian stretch/non-free armhf Packages
     release o=Raspbian,a=stable,n=stretch,l=Raspbian,c=non-free,b=armhf
     origin mirrordirector.raspbian.org
 500 http://mirrordirector.raspbian.org/raspbian stretch/contrib armhf Packages
     release o=Raspbian,a=stable,n=stretch,l=Raspbian,c=contrib,b=armhf
     origin mirrordirector.raspbian.org
 500 http://mirrordirector.raspbian.org/raspbian stretch/main armhf Packages
     release o=Raspbian,a=stable,n=stretch,l=Raspbian,c=main,b=armhf
     origin mirrordirector.raspbian.org
Pinned packages:
```

It seems that just installing "security" updates may not be possible. The best solution considering the available repositories might be installing all updates from stable repositories.

markushuber, 2017-10-04:
Great catch @tobiasdam . It seems that the raspbian/raspberrypi projects do not maintain security-labels. It seems that with the current stretch image, all important security updates come from the *archive.raspberrypi.org* which are labeled as *testing*.

```
root@upribox:/etc/apt/sources.list.d# unattended-upgrade -d
Initial blacklisted packages: 
Initial whitelisted packages: 
Starting unattended upgrades script
Allowed origins are: ['origin=Debian,codename=stretch,label=Debian-Security']
Checking: bluez ([<Origin component:'main' archive:'testing' origin:'Raspberry Pi Foundation' label:'Raspberry Pi Foundation' site:'archive.raspberrypi.org' isTrusted:True>])
Checking: dhcpcd5 ([<Origin component:'main' archive:'testing' origin:'Raspberry Pi Foundation' label:'Raspberry Pi Foundation' site:'archive.raspberrypi.org' isTrusted:True>])
Checking: dnsmasq ([<Origin component:'main' archive:'testing' origin:'Raspberry Pi Foundation' label:'Raspberry Pi Foundation' site:'archive.raspberrypi.org' isTrusted:True>])
Checking: dnsmasq-base ([<Origin component:'main' archive:'testing' origin:'Raspberry Pi Foundation' label:'Raspberry Pi Foundation' site:'archive.raspberrypi.org' isTrusted:True>])
Checking: git ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'mirrordirector.raspbian.org' isTrusted:True>])
Checking: git-man ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'mirrordirector.raspbian.org' isTrusted:True>])
Checking: libidn2-0 ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'mirrordirector.raspbian.org' isTrusted:True>])
Checking: libperl5.24 ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'mirrordirector.raspbian.org' isTrusted:True>])
Checking: libwbclient0 ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'mirrordirector.raspbian.org' isTrusted:True>])
Checking: linux-libc-dev ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'mirrordirector.raspbian.org' isTrusted:True>])
Checking: perl ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'mirrordirector.raspbian.org' isTrusted:True>])
Checking: perl-base ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'mirrordirector.raspbian.org' isTrusted:True>])
Checking: perl-modules-5.24 ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'mirrordirector.raspbian.org' isTrusted:True>])
Checking: python-pip ([<Origin component:'main' archive:'testing' origin:'Raspberry Pi Foundation' label:'Raspberry Pi Foundation' site:'archive.raspberrypi.org' isTrusted:True>])
Checking: python-pip-whl ([<Origin component:'main' archive:'testing' origin:'Raspberry Pi Foundation' label:'Raspberry Pi Foundation' site:'archive.raspberrypi.org' isTrusted:True>])
Checking: raspi-config ([<Origin component:'main' archive:'testing' origin:'Raspberry Pi Foundation' label:'Raspberry Pi Foundation' site:'archive.raspberrypi.org' isTrusted:True>])
Checking: raspi-copies-and-fills ([<Origin component:'main' archive:'testing' origin:'Raspberry Pi Foundation' label:'Raspberry Pi Foundation' site:'archive.raspberrypi.org' isTrusted:True>])
Checking: samba-common ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'mirrordirector.raspbian.org' isTrusted:True>])
pkgs that look like they should be upgraded: 
Fetched 0 B in 0s (0 B/s)                                                                                                                                                                                         
fetch.run() result: 0
blacklist: []
whitelist: []
No packages found that can be upgraded unattended and no pending auto-removals
```

tobiasdam, 2017-10-05:
After doing some investigation, the situation seems to be as follows:

1. I checked the "testing" repository online (http://archive.raspberrypi.org/debian) and found that the stretch archive is still marked as testing for whatever reason. That means the testing repo is not a real debian buster testing repo, but just a mislabeled stretch stable.

2. Both the archive and the mirrordirector repository seem to receive security updates. For example dnsmasq version `2.76-5+deb9u1`, which contains fixes for several vulnerabilities (https://www.debian.org/security/2017/dsa-3989), is contained in both repos.

```
root@upribox:/home/upri# apt-cache showpkg dnsmasq                                                                                                                                                                   
Package: dnsmasq                                                                                                                                                                                                     
Versions:                                                                                                                                                                                                            
2.76-5+rpt1+deb9u1 (/var/lib/apt/lists/archive.raspberrypi.org_debian_dists_stretch_main_binary-armhf_Packages)                                                                                                      
 Description Language:                                                                                                                                                                                               
                 File: /var/lib/apt/lists/mirrordirector.raspbian.org_raspbian_dists_stretch_main_binary-armhf_Packages                                                                                              
                  MD5: d4d0c88134cc703150a59b83cb68304c                                                                                                                                                              
 Description Language:                                                                                                                                                                                               
                 File: /var/lib/apt/lists/archive.raspberrypi.org_debian_dists_stretch_main_binary-armhf_Packages                                                                                                    
                  MD5: d4d0c88134cc703150a59b83cb68304c                                                                                                                                                              
 Description Language:                                                                                                                                                                                               
                 File: /var/lib/dpkg/status                                                                                                                                                                          
                  MD5: d4d0c88134cc703150a59b83cb68304c                                                                                                                                                              
                                                                                                                                                                                                                     
2.76-5+rpi3 (/var/lib/dpkg/status)                                                                                                                                                                                   
 Description Language:                                                                                                                                                                                               
                 File: /var/lib/apt/lists/mirrordirector.raspbian.org_raspbian_dists_stretch_main_binary-armhf_Packages                                                                                              
                  MD5: d4d0c88134cc703150a59b83cb68304c                                                                                                                                                              
 Description Language:                                                                                                                                                                                               
                 File: /var/lib/apt/lists/archive.raspberrypi.org_debian_dists_stretch_main_binary-armhf_Packages
                  MD5: d4d0c88134cc703150a59b83cb68304c
 Description Language:
                 File: /var/lib/dpkg/status
                  MD5: d4d0c88134cc703150a59b83cb68304c

2.76-5+deb9u1 (/var/lib/apt/lists/mirrordirector.raspbian.org_raspbian_dists_stretch_main_binary-armhf_Packages)
 Description Language:
                 File: /var/lib/apt/lists/mirrordirector.raspbian.org_raspbian_dists_stretch_main_binary-armhf_Packages
                  MD5: d4d0c88134cc703150a59b83cb68304c
 Description Language:
                 File: /var/lib/apt/lists/archive.raspberrypi.org_debian_dists_stretch_main_binary-armhf_Packages
                  MD5: d4d0c88134cc703150a59b83cb68304c
 Description Language:
[... cut ...]
4. Since both repository contain "security" updates and it is not predictable which repo is needed, adding both repositories to `50unattended-uprgades` and also some addtional lines to skip update prompts (also needed for the dnsmasq update), seems to be the only possible solution.

```
Unattended-Upgrade::Origins-Pattern {
        //"origin=Debian,codename=${distro_codename},label=Debian-Security";
        "o=Raspbian,a=stable,label=Raspbian";
        "o=Raspberry Pi Foundation,a=testing,n=stretch,label=Raspberry Pi Foundation";
};

Dpkg::Options {
     "--force-confdef";
     "--force-confold";
};
```

5. This solution would result in effectively installing every update.

markushuber, 2017-10-05:
Solution confirmed, please go ahead and add both stretch repos to unattended upgrades.

closed 2017-10-25 by tobiasdam