usableprivacy/upribox #87: Static IP: requires reboot and hard to use

markushuber, 2017-09-27:
Setting a static IP requires a reboot of the Raspberry Pi in order to use the static IP. Testing on a stretch-based v2 image.

The usability needs to be improved:
- If static settings are saved successfully, the static IP feature should be automatically enabled. The user should furthermore be redirected to the settings page after successful setting a static network configuration.

markushuber, 2017-10-04:
This issue requires additional testing.

christianbrandl, 2017-11-14:
There is no reboot required anymore. After the static ip is set, the static ip feature will be automatically enabled. In addition there are no redirects. You stay on the configuration page, even after the tasks are finished. The previous ip is active til the next reboot. Afterwards only the static ip remains.

closed 2017-11-14 by christianbrandl