usableprivacy/upribox #92: Device vanishes from device overview

tobiasdam, 2017-10-05:
Currently, devices, that have not been determined online by `registrar` or `parse_user_agents` for more than one week, are not shown in the device overview.

If a device has a static IP (registrar won't work) and is configured to use "no mode" (parse_user_agents won't work). it appears to be offline forever.

markushuber, 2017-11-08:
Any progress on this bug. @is151006 can you verify if the issue still exists?

markushuber, 2017-11-10:
cronjob to probe for offline / no-mode devices every 1h.

ThorstenS-linux, 2017-11-11:
to scan my network, I using this:
```
fping -g 192.168.2.0/24 -Q 3 -a
```
faster is this 
``
nmap -TAggressive -n -sS -p80 192.168.2.1 192.168.2.254 &> /dev/null
arp -a | fping -AaQ3
```

tobiasdam, 2017-11-17:
fixed by  5ad79746b9bd30ff2a638b6a36835755836c1d3

closed 2017-11-17 by tobiasdam