usableprivacy/upribox #96: Enable autoclean for unattended upgrades

markushuber, 2017-11-07:
`/var/cache/apt/`keeps growing due to the auto updates we perform. Configure unattended upgrades to perform cleanups every 7 days.

ThorstenS-linux, 2017-11-07:
in `roles/unattended_upgrades/templates/02periodic` you have this:
```
// Do "apt-get autoclean" every n-days (0=disable)
APT::Periodic::AutocleanInterval "{{  default_settings.unattended_upgrades.general.autoclean_days if not (ansible_local is defined and ansible_local.unattended_upgrade is defined and ansible_local.unattended_upgrade.general is defined) else ansible_local.unattended_upgrade.general.autoclean_days| default( default_settings.unattended_upgrades.general.autoclean_days) }}";  
```
So I think all you have to do is set `default_settings.unattended_upgrades.general.autoclean_days` to 7

ThorstenS-linux, 2017-11-07:
I read `man apt.conf` and found under PERIODIC AND ARCHIVES OPTIONS a reference to `/usr/lib/apt/apt.systemd.daily`.
There I found this:
```
#  APT::Periodic::AutocleanInterval "0";
#  - Do "apt-get autoclean" every n-days (0=disable)
#
#  APT::Periodic::CleanInterval "0";
#  - Do "apt-get clean" every n-days (0=disable)
```
so you have to implement `APT::Periodic::CleanInterval` to run `apt clean` every X days.

tobiasdam, 2017-11-08:
Since we do not need cached packages most of the time, I chose the `CleanInterval` option.
@ThorstenS-linux thank you for your help!

closed 2017-11-08 by tobiasdam