usableprivacy/upribox #101: Web app login

FirstNerd, 2017-11-11:
Fresh install of version 1 on pi 3 B, only silent mode active.  Try to login to web app and I get a server error 500.

markushuber, 2017-11-11:
Hi @FirstNerd, we will investigate the issue - thanks for reporting it.

FirstNerd, 2017-11-12:
[image]
[image]

screenshots of me able to ssh into the pi3 while running upribox v1 img 
with nginx status and nginx error log.. if anymore info is needed let me know I'm glad to help

markushuber, 2017-11-12:
Thanks for reporting this issue @FirstNerd. I triaged the problem and committed a fix (redis-server was failing due to an inaccessible logfile). Update should reach your box withing four hours.
Please let us know if the commit fixed your issue.

FirstNerd, 2017-11-12:
anyway I can force an update via ssh terminal to check if the issue been resolved?

FirstNerd, 2017-11-13:
Issue resolved....

closed 2017-11-13 by FirstNerd