usableprivacy/upribox #107: Upribox V1  TLS Error using Win10

dickdevries, 2017-11-18:
Hi,
I have an issue running Upribox V1 vpn from 2 windows 10 laptops. Both are on Win10 v 1703 build 15063.726.
On both laptops I installed OpenVPN v. 2.4.4 and can connect PiVPN with 2048 bit key without a problem.
On both laptops I get an error if I try to connect to Upribox V1.
I also use a few android devices and they connect to Upribox without any problem. (they also connect to the Pivpn without issues).

This is the logfile:
[upribox-dick.log](https://github.com/usableprivacy/upribox/files/1484698/upribox-dick.log)

Advice is very appreciated.
I already did a reinstall of OpenVPN client on one of the laptops, but no difference. 
Dick

dickdevries, 2017-11-18:
Some additional information:
I just installed OpenVPN on a Windows 7 workstation and created a new key.
The connection to the Upribox is working fine, without an issue at all.
here is the logfile:
[upribox-nicole.log](https://github.com/usableprivacy/upribox/files/1484936/upribox-nicole.log)

dickdevries, 2017-11-19:
Additional information:
If I connect with the Windows 10 Laptop from a different lan, I am able to connect Upribox without a problem. With my Android devices I can connect Upribox in my LAN, but Windows 10 client can not.
I am happy that I can connect with windows 10 client when I am travelling and use the cellphone net or an open wifi access point. Weird that the Windows Client is not connecting via the lan.
The router in my lan is a fritzbox 7390.

I don't see a reason to believe that it is a bug in the Upribox code.

tobiasdam, 2017-11-20:
Hey Dick,

this seems to be some really weird client-side connection problem.
Since you are able to connect to the upribox from inside your LAN with your android phone and from the Internet with your Win 10 Laptops, I don't think this issue is network-related.

I suggest reviewing configuration that might depend on your location, e.g.:
- firewall settings, if you use a third-party firewall (or anti-virus with integrated firewall)
- windows firewall settings with the public/private network type in mind

For debugging purposes you could try creating a new openvpn profile using the internal ip address of the upribox as domain. Then try connecting from inside your LAN to the upribox with your Win 10 computer.

dickdevries, 2017-11-20:
Hi Tobias,

 

Thanks for your debugging suggestion.

I did create a new profile with internal LAN address and it is working fine, without any problem.

 

The fact is that the same Laptops can connect within my LAN with the same OpenVPN client and the same firewall settings my PiVPN without a problem. 

So that makes me think the reason is in the Upribox.

tobiasdam, 2017-11-22:
Since you are able to connect to the upribox with your Android device and your Windows 7 device inside your LAN and are able to connect with your Windows 10 devices from the Internet, I do not think that this is a configuration issue of the upribox. 

If you could provide us your openvpn logfile from the upribox, we could further debug this issue and check whether this is an upribox issue or not.

You would need to log into your upribox via `ssh` and copy the logile `/var/tmp/log/openvpn.log`.

dickdevries, 2017-11-23:
I was not able to test win7 over lan. I will list what I was able to test with Upribox:
For the test below I created keys with my domainname (vriXXXXXX.no-ip.biz)
Android over LAN is OK
Android over Internet is OK
Windows 7 over LAN  ( not able to test, I have no local Windows 7 only on a remote location )
Windows 7 over Internet is OK
Windows 10 over LAN is not connecting
Windows 10 over Internet is OK

I generated a new key, based on the LAN IP address of the Upribox and with that key I was able to connect my windows 10 laptops with Upribox over LAN. 

I did the same test with the same clients on the same raspberry, switched SD card to run PiVPN:
For Android, Windows7 and Windows 10 it is possible to connect over the Internet and over LAN. 

If it helps to debug I will sent copy the my openvpn.log file, after I did a reboot of Upribox and a few connects with the win10 client over LAN with the domain Key.

dickdevries, 2017-11-23:
I looked at the openvpn.log file and think it is recycled this morning, so I made a successful connect and a not successful connect and copied the log file.
[openvpn.log](https://github.com/usableprivacy/upribox/files/1498998/openvpn.log)

hope this will help.

dickdevries, 2017-11-24:
additional test and info:
copied openvpn.log this morning after 2 connect attempts with win10 over lan, both failled.
this is showing in the connect form:
[image]
 Nothing is added to the logfile, so if the connection fails no entries are made in the openvpn.log file.
[openvpn.log](https://github.com/usableprivacy/upribox/files/1501449/openvpn.log)

After that I connected win10 over LAN with the profile where the LAN ip is in the domain name. connection is OK now and also information is added to openvpn.log file
[openvpn.log](https://github.com/usableprivacy/upribox/files/1501464/openvpn.log)

hope you can use the info.

tobiasdam, 2017-11-24:
Thank you for providing your logfiles!

Because your upribox logiles do not contain any information in case the connection attempt fails, it seems that your clients are not able establish a network connection to the upribox under these circumstances. Hence, this does not seem to be an issue of the upribox.

Some routers are configured in a way, that does not allow to connect to service inside your network when specifiying the outbound IP address. But such routing or NAT issues are unlikely to be the problem, because you are able to connect inside your LAN with your Android phone. Therefore this might be a problem of the clients.

dickdevries, 2017-11-25:
I don't understand why I can connect PiVPN over LAN with the same win10 clients, in te same network environment, using the same Openvpn clients and using the same Raspberry Pi (running PiVPN at that time, using 2048 bit encryption key profiles).
This should mean that the Openvpn client is compatible with PiVPN over LAN and incompatible with Upribox over LAN ?

This made me think of an Upribox issue.

markushuber, 2018-10-18:
No new updates regarding this issue, closing it for now.

closed 2018-10-18 by markushuber