usableprivacy/upribox #109: Leichte Inkonsistenzen bei Schriftgrößen

chrisi092, 2017-11-26:
Generell wurden bei der Papieranleitung (basierend auf Version auf der Dropbox Version "upribox2-Anleitung")  einige Inkonsistenzen bei den Schriftgrößen bei Überschriften auf gleicher Hierarchie-Ebene festgestellt. Hier sollte nochmal kontrolliert und nachgebessert werden.

christianbrandl, 2017-12-01:
Was meinst du mit Überschriften auf gleicher Hierachie-Ebene? Soweit wie ich das sehe gibt es pro Segment nur eine Überschrift oder meinst du das Überschriftsegment für die ganze Seite (türkise Kästchen)?

chrisi092, 2017-12-01:
Grundsätzlich ging es nur darum, dass in der Version des Styleguides aufgrund der ich bewertet hatte manche Überschriften (z.B. H2 o.ä.) auf einer Seite 17 pt und auf der anderen 18 pt groß waren obwohl sie eigentlich die gleiche Hierarchie-Ebene abbilden und identisch sein sollten.

christianbrandl, 2017-12-01:
Die Indesign-Anleitung wird nicht mehr verwendet und in der neuen Latex Version sind Größen allgemein festgelegt, daher verwenden beide Seiten die gleichen Schriftgrößen.

closed 2017-12-01 by christianbrandl