usableprivacy/upribox #122: Server Error (500) V2

dickdevries, 2018-01-13:
Hi,

This morning I wanted to check statistics.
After logging in on the https server (https://upri.box:4300), my browser is showing me the a blanc page with Server Error (500).
For the http server it is the same
I used a few different browsers on different platforms, all do the same.
I am able to access the vpn.
a reboot did not change anything
The Upribox version is V2.

Dick

tobiasdam, 2018-01-16:
Hey,

could you please provide the logfile `/var/tmp/log/django_error.log` from the upribox, after you retry accessing the page to create errors in the logfile? 

Please ensure the upribox was up and running for at least one hour, so that all new updates have been applied.

tobiasdam, 2018-01-17:
After logging in via ssh you need to use `sudo cp /var/tmp/log/django_error.log /home/upri/django_error.log` to copy the file into the home directory and change the ownership of the copy via `sudo chown upri:upri /home/upri/django_error.log` to the `upri` user. Afterwards you can retrieve the file from the upribox using `scp` or other means.

Alternatively, you can enter `sudo su` after logging in to become the root user, which enables you to view the file and access the log directory.

dickdevries, 2018-01-17:
Thanks, I used sudo su, but don't see the file django_error.log

[image]

tobiasdam, 2018-01-17:
This might mean the last updates were not applied or failed for some reason.
Please check your upribox version with `git -C /var/lib/ansible/local/ rev-parse --short HEAD` and additionally provide the file `var/tmp/log/ansible-pull.log`.

You can also try to initiate the update process yourself via `sudo ansible-cron &`. The update process is running in the background and might take some time to finish. You should see an entry inside `ansible-pull.log` with an according timestamp.

dickdevries, 2018-01-17:
On the version check the system returns : 7823370

and the ansible-pull.log file
[ansible-pull.log](https://github.com/usableprivacy/upribox/files/1639456/ansible-pull.log)

I did not try to initiate the update process now.

tobiasdam, 2018-01-18:
It seems that the new update was pulled, but it has not been successfully applied by the time you copied the logfile.

The system was currently trying to apply the changes for the second time, after the first run failed to update the system packages (apt cache). This might have been caused by a network error or by a bug of the ansible apt module.

dickdevries, 2018-01-18:
Additional information:

I was under the impression, since a few weeks, that the upribox caused frozen television streams. Not permanent but now and then 5 - 10 seconds. To verify this I disconnected the network cable from the upribox while looking streams. I never had a frozen stream when the upribox was disconnected from the network. So the power was on all the time. For the rest of the time, at least 23 hours a day, the upribox is connected to the network.

dickdevries, 2018-04-14:
Hi Tobias,

 

My issue didn’t get resolved.

A couple of weeks ago I did a complete new setup by copying the Upri v2 image to the sdcard.

It was working fine for the last 3 weeks. Today I found out that I was unable to access via HTTP and HTTPS again.

This time the login screen is showing but my login and password is not accepted.

I am able to get in via SSH. Wlan is running and I am able to connect the Wlan.

 

Regards,  Dick