usableprivacy/upribox #123: Login problem V2

dickdevries, 2018-04-17:
Since last week I am no longer able to login upribox via the web interface. The web interface is running but doesn't accept my login name or Password since a couple of days. All the functions (Wlan, VPN, SSH, HTTP, HTTPS) are running. I am able to login via SSH. I see a lot of data traffic to the upribox IP address. 
Is my upribox hacked? is there a way to see If a login / password change is made?
This is the second time something curious is going on with my upribox. In January I had to start from scratch with a clean image. No problem, that is what testing is for, but if someone is using a 'backdoor' in the upribox software to get into my network I am not happy.

markushuber, 2018-04-27:
Hi @dickdevries,

>   I am no longer able to login upribox via the web interface.

Great that you changed the default password and there is no straightforward way to change it if you forgot it for security reasons.

> I see a lot of data traffic to the upribox IP address

What do you use to measure the traffic? In automatic mode, the traffic of all your devices should be routed through your box - so this is expected behavior.

> if someone is using a 'backdoor' in the upribox software to get into my network I am not happy.

We would not be happy either and there is no 'backdoor' - thats why we open source our software.

dickdevries, 2018-04-27:
Hi,

 

Being an IT pro / software developer I know the passwords I used in the Upribox.

 

I monitored the traffic using ‘Wireshark’ looking at the specific IP address of the upribox, knowing that the traffic is not initiated by one of the local workstations.

markushuber, 2018-05-03:
Hi @dickdevries ,

Glad to hear that you are an IT pro. I spent some time to try to reproduce your login issue.
Could you please run the following command via `SSH`: `sudo /usr/local/bin/ansible-cron --force`
Once the command is completed (it reruns the entire software setup) please verify that there were no errors: `sudo tail /var/tmp/log/ansible-pull.log`, the last iine of the output should look similar to

> 127.0.0.1                  : ok=281  changed=26   unreachable=0    failed=0

If there were no errors (`failed=0`), please try to login with the password you set. If this does not help or produces errors (failed>0) let us know.