usableprivacy/upribox #130: Broken ntopng repository

tobiasdam, 2018-05-02:
Apparently, the [ntopng apt-stable repository](http://packages.ntop.org/apt-stable/stretch_pi/) is broken and does not contain any packages at the moment. As a result, apt will install the old 2.4 version

Adding `http://packages.ntop.org/apt/stretch_pi/`  to the apt sources might work as hotfix, but this will result in installing development versions of ntopng.

markushuber, 2018-05-03:
Thank you for spotting the issue with ntopng, is there already an issue at their github page - we can reference / watch?

tobiasdam, 2018-05-03:
I created a workaround using the development repository (which only contains a stable version build at the moment...). The tasks retrieve the latest stable version in the ntopng github repo and install that specific version searching in the ntopng apt (dev), ntopng apt-stable and the raspbian repo.

tobiasdam, 2018-05-03:
The new issue is at https://github.com/ntop/ntopng/issues/1767

markushuber, 2018-05-17:
Packages were unavailable again and reported to ntopng maintainers https://github.com/ntop/ntopng/issues/1780
Finally, opted for compiling and creating deb packages from the ntopng 3.2-stable branch and hosting the packages on upribox.org

commit 761fc3a9 (2018-05-03): Merge pull request #132 from usableprivacy/ntopng_fix

workaround for broken ntopng repositories closes #130
closed 2018-05-17 by markushuber