usableprivacy/upribox #137: Red LED turns off on Raspberry Pi 3B

niansa, 2018-07-30:
The red LED of my Raspberry Pi 3B turns off some timer after i booted upribox up. Averything seems to work fine, on HDMI i see no lightning bolt or rainbow box, but the Raspberry Pi gets hot very quickly. After shutdown the red LED turns on again.

markushuber, 2018-09-20:
We switch off the red light on purpose to indicate once the upribox finished startup (constant green light). Regarding your Pi's heat, which Pi are you using? We have no reports of overheating due to our Software yet.

markushuber, 2018-10-16:
Raspberry Pi 3B should work fine with our software. On initial boot of a new upribox installation, strong unique cryptographic certificates are generated - this might have caused the heat you noticed.
Closing this ticket for now, if you notice heat issues during day-to-day usage of your upribox - please let us know.

closed 2018-10-16 by markushuber