usableprivacy/upribox #138: 802.11ac @ 80MHz AP mode

seurat-atreides, 2018-09-17:
Hi,
Is it possible to configure hostapd to operate 802.11ac @ 80MHz?
If so what parameters must I include in the hostapd.conf file?

markushuber, 2018-09-20:
Hi @seurat-atreides 
We did not implement an option yet to switch the WiFi to AC in 80MHz mode. Would be glad if you report working settings in hostapd.conf back to us, so we can implement an option to switch to AC mode (local changes to hostapd will eventually get overridden by Software updates).

tobiasdam, 2018-10-19:
802.11ac is only supported by the new Raspberry Pi 3 B+ and there seems to be a [bug](https://github.com/raspberrypi/linux/issues/2619) preventing AC for every country_code other than `country_code=US`

seurat-atreides, 2018-10-25:
Hi Tobias,
Not true, I'm currently running 802.11ac on an RPI 3 B+ with 40MHz bandwidth for country_code=DE.

tobiasdam, 2018-10-25:
Since you explicitly stated "80MHz" in the title, I did not mention it in my comment above. The issue also states "5Ghz" (80MHz channel width) in the title and it belongs to an official raspberrypi repository. Even if it would be working for you, it is certainly not working for some others.

seurat-atreides, 2019-02-15:
Hi *,
for those still interested in getting their Upribox to work as an AP with 80MHz BW.
I did some more testing and got the WLAN IF to work in AP mode with 80MHz BW.
Here is the hostapd.conf file I used:
[hostapd_5G.conf.txt](https://github.com/usableprivacy/upribox/files/2868925/hostapd_5G.conf.txt)

Here is a screenshot from my smartphone:
[image]
I had to set the country_code=US instead of DE to get this to work! 
BTW: I removed the metal case cover to get better RF power.
I will be buying a new plastic case.

closed 2019-02-15 by seurat-atreides