usableprivacy/upribox #147: WLAN: hostapd.conf persistent changes

seurat-atreides, 2019-02-15:
Hi *,

I have come up with a hostapd.conf file that will allow me to run in AP mode with 80MHz BW: https://github.com/usableprivacy/upribox/issues/138
The AP is working fine with country_code=US. My problem is that every time I make some change to the WLAN in the Web-GUI the hostpd.conf file gets overwritten.
Where do I find the template that changes this file?

seurat-atreides, 2019-02-18:
Hi *,
I've found the location of the template responsible for populating the hostapd.conf file:
>/var/lib/ansible/local/roles/wlan/templates/hostapd.j2

I have modified this file to have the following content:
```
interface=wlan0
bssid={{ ansible_wlan0.macaddress | regex_replace('^..', magic_mac_prefix) | regex_replace('..$', magic_mac_postfix) }}
channel=36
hw_mode=a
ieee80211n=1
require_ht=1
ieee80211ac=1
require_vht=1
ieee80211d=1
ieee80211h=0
ht_capab=[HT40+][SHORT-GI-20][SHORT-GI-40][DSSS_CK-40][MAX-AMSDU-3839]
vht_capab=[MAX-MPDU-3895][SHORT-GI-80][SU-BEAMFORMEE]
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=42

#ssid
ssid={{ default_settings.wlan.upri.ssid if not (ansible_local is defined and ansible_local.wlan is defined and ansible_local.wlan.upri is defined) else ansible_local.wlan.upri.ssid | default(default_settings.wlan.upri.ssid) }}
country_code=US
auth_algs=1
wpa=2
macaddr_acl=0
ignore_broadcast_ssid=0
wpa_passphrase={{ default_settings.wlan.upri.passwd if not (ansible_local is defined and ansible_local.wlan is defined and ansible_local.wlan.upri is defined) else ansible_local.wlan.upri.passwd | default(default_settings.wlan.upri.passwd) }}
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP
```
After changing this file you must activate or modify the WiFi settings to get the AP to start in 802.11ac with 80MHz BW.

markushuber, 2019-02-22:
Hi @seurat-atreides,
Thank for reporting a working 5Ghz config file back to us. Yes all changes to config files get overwritten once there are new commits or if you use the GUI.
We basically have to implement an option to set your alternative config in the template. Will implement this in the next week and keep you posted.