usableprivacy/upribox #148: Static IP configuration, Server Error (500) , upribox 2.0

intrepid-ibex, 2019-06-28:
When I click Static IP configuration, I get Server Error (500)
What happens?

markushuber, 2019-07-09:
Hi @intrepid-ibex unfortunately I could not reproduce the issue you experienced with static IP settings in the web interface.

Do you still have the issue?
It would be helpful if you could provide debug information via Email: https://upribox.org/kontakt/
Debug information required: latest backup from `/var/upribox-backups/weekly`

intrepid-ibex, 2019-10-07:
Hi, when I try to access the weekly folder , i got "permission denied".
On my PI the green LED is always ON, and the red LED is always OUT.
Is this correct?
Best regards.

markushuber, 2019-10-10:
Once the upribox has completed the startup, the green LED will be always on while the red LED is out - this is expected.
The backup files are protected from access via a standard user, run `sudo su` to become root before accessing the backup-files.

closed 2019-12-06 by markushuber